networkupstools/nut #3092: upssched no longer passes the NOTIFYTYPE and UPSNAME environment variables into cmdscript

kennethso168, 2025-09-16:
version 2.8.4

I believe this is caused by b75bea251a7f8b6f880a0d63d306532ad9250065

Previously I have these variables in another deployment with version 2.8.1

The example [`upssched-cmd`](https://github.com/networkupstools/nut/blob/master/clients/upssched-cmd) script still references these two variables, so I would expect these two variables to be available

I suggest either pass the environment variables into the script or modify the example script. Thank you.

jimklimov, 2025-09-17:
But it seems the premise of your question is right: even if `upssched` running as a timer daemon should not keep those envvars and impose them on all CMD runs it launches when timers expire, it should somehow learn and keep these variables from when each timer was started by some line like
```
AT ONLINE * START-TIMER LINE-STATUS-10 600
```
...and `setenv()` them after forking to start a CMD.

jimklimov, 2025-09-17:
Not sure what the NOTIFYTYPE should be here though. Original (if possible/available)? An explicit `TIMER`? Some combo of the two (e.g. `ONLINE-TIMER`)?

jimklimov, 2025-09-17:
Actually "what is a valid UPSNAME in a timer?" is a trickier question. Consider you have many UPSes and a big outage, and many hits to `AT ONBATT *` start the same timer. When it elapses, what do you put into UPSNAME wjen calling the script? A random one? A list of all who started it?

jimklimov, 2025-09-17:
Likewise, not sure if it is invalid to have several config lines starting the same timer. Maybe due to different trigger events even.

```
AT CAL * START-TIMER echo-no-worry 10
AT ECO * START-TIMER echo-no-worry 10
```

IF so, and if both did start the timer before it elapsed to run unambiguously, which one event goes to NOTIFYTYPE?

jimklimov, 2025-09-21:
Ok, the task gets sort of easier: per `start_timer()` it seems we get a new timer (not an addition to existing one) if "name" is same as something started before. So each tracked event has unique `UPSNAME` and `NOTIFYTYPE` values, but many of these can fire in a short succession if something large happens - e.g. many UPSes go "ONBATT", and after a timeout of them being in this state it gets handled for each one individually, without knowledge about others unless your script is coded to know about that.

I guess a "SHARED_TIMER" can be introduced for cases where sysadmins want a common single-run handler for possibly numerous devices and/or notified events.