networkupstools/nut #3099: packaging scripts and others like upsdrvsvcctl/nut-driver-enumerator.sh.in should use the EGREP they located during configure/build

nbriggs, 2025-09-26:
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in contains grep invocations like
```
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in:if set | grep -E '^(shell|version|t?csh)' | grep -E 't?csh' >/dev/null ; then
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in:        echo "$SVCINST_PROPS" | grep -E "^${SVCINST}${TABCHAR}"'$' >/dev/null && echo "$SVCINST"
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in:        echo "$SVCINST_PROPS" | grep -E "^${SVCINST}${TABCHAR}" >/dev/null || echo "$SVCINST"
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in:        echo "$SVCINSTS" | grep -E "^${SVCINST_PROP}"'$' >/dev/null || echo "$SVCINST_PROP"
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in:        echo "$SVCINST_PROPS" | grep -E "^${SVCINST}${TABCHAR}"'$' >/dev/null && echo "$SVCINST"
scripts/upsdrvsvcctl/nut-driver-enumerator.sh.in:        echo "$SVCINST_PROPS" | grep -E "^${SVCINST}${TABCHAR}" >/dev/null || echo "$SVCINST"
```
However it has gone to some trouble during configure step to determine what the functional names of EGREP, FGREP, and GREP are -- and on a Solaris system those are
```
scripts/upsdrvsvcctl/Makefile:EGREP = /usr/gnu/bin/grep -E
scripts/upsdrvsvcctl/Makefile:FGREP = /usr/gnu/bin/grep -F
scripts/upsdrvsvcctl/Makefile:GREP = /usr/gnu/bin/grep
```
not just plain `grep`.  So... it would be helpful if the transformation from nut-driver-enumerator.sh.in to nut-driver-enumerator.sh took this into account.

jimklimov, 2025-09-26:
Thanks! (First reaction) The emoticons of GitHub sorely lack a "facepalm" icon :-D

On one hand, this issue may be seen as a follow-up from #1660 to use `grep -E` because some systems issued warnings at any `egrep` shorthand use - however (Second reaction - looking at code) "going through trouble" seems to be part of automake core, not much noted consciously. There is one `AC_PROG_EGREP` in `configure.ac` (since the beginning of time, 2005 or before), and a few mentions that arguably disregard it in `docs/Makefile.am` and `docs/man/Makefile.am` by defining `EGREP = grep -E` (those are mine, likely from that ticket). 

The uppercase variables are only `$(EGREP)`'s used in those two Makefiles, and a `$(GREP)` once in `scripts/python/Makefile.am`, and a bit from before such detection in `ci_build.sh`.

In short, such automake-detected variables are practically not used in written code at the moment. Same for `SED` and `AWK` by the way.

jimklimov, 2025-09-26:
BTW, does the current situation cause any practical issues? Is there some incompatibility with the `grep` etc. implementation that it finds in `PATH` (arguably *that* would be a bug - depending on particularly GNU grep etc.)

I was running NUT builds on Solaris 10, OpenSolaris and later illumos distros, installed via old-style `make package` artifacts, and there were no hiccups noted, especially with NDE scripts (get a lot of occasional testing in such loops as part of pre-/post- install/remove scripts of the package).

nbriggs, 2025-09-26:
Yeah, unfortunately it does cause a problem - `/usr/gnu/bin` is not on root's path (for various reasons) and `-E` is not an option accepted by Solaris' `/usr/bin/grep` --
```
grep -E test /tmp/test
grep: illegal option -- E
Usage: grep [-c|-l|-q] -bhinsvw pattern file . . .
```
I have `/usr/gnu/bin` on the path while I'm building and packaging it, so `grep -E` is ok, but after installing, at runtime, the script `scripts/upsdrvsvcctl/nut-driver-enumerator.sh` fails because it has the standard root PATH.

My. workaround was to go through all the scripts that it installed and change all `grep` to `/usr/gnu/bin/grep`.  Which is fine for now, but not a desirable long-term solution, hence the issue.

alanc, 2025-09-26:
On Solaris, your choices for egrep are:
- `/usr/bin/egrep` - available on all releases since Solaris 1.0 in the early 1990's
- `/usr/xpg4/bin/grep -E` - available on Solaris 2.5 & later
- `/usr/gnu/bin/grep -E` or `/usr/gnu/bin/egrep` - available on Solaris 11.0 & later

When building FOSS for Solaris, we certainly end up putting `/usr/gnu/bin` first in $PATH quite often in order to deal with scripts that expect support for `grep -E`, GNU extensions to gettext, etc.

jimklimov, 2025-09-26:
Thanks Alan. On OpenIndiana and OmniOS I see `/usr/bin/grep` and `/usr/xpg4/bin/grep` as one being a symlink to the to the other, so wondered if in proper Solaris it was a third program or also a convenience link to one of the implementations.

jimklimov, 2025-09-28:
Thanks! A PR would be most welcome :)

nbriggs, 2025-09-28:
Yeah, I can do that.

nbriggs, 2025-10-01:
I am going to rework some of the changes I was going to propose.  I believe that it is necessary to have GNU tools on the PATH while building on Solaris, and that there's no simple mechanism to ensure that all the m4 macros/etc that use "grep" and "grep -E" that end up in the configure script itself are automatically located.

I will make sure that the autogen.sh script's use of `find` is compatible with Solaris (and more modern systems), that Makefiles generated from Makefile.am input use the correct programs (grep, grep -E, awk, sed), and that scripts that get incorporated into the Solaris SYSV package use the correct programs found during the configuration step.

jimklimov, 2025-10-01:
Tried to re-run a build on my Solaris 8 worker. Apparently I haven't tried for over a year, and/or was not too attentive to logged messages as long as the scripts claimed to succeed... So ended up also fixing some parts, hopefully not overlapping with yours seen so far.

jimklimov, 2025-10-01:
At least, I can cherry-pick your https://github.com/networkupstools/nut/commit/da812cca653b58229addc9293083fb5bad549d60 commit into that branch of mine without conflicts :)

jimklimov, 2025-10-01:
Actually it seems sufficient to detect (E)GREP early in `configure.ac` (like, just after `AC_INIT` to be sure), and use the variables in the `ac` script and m4 files. At least, `grep -w grep configure` has very few hits and they seem to be from autoconf internals.

At least with "tgcware" GNU tools in path, config and build (including `make -ksj`) now passed without reporting errors on this side (there are some concerns listed below: about missing methods, and platform-specific about limited range type in comparisons - pragmas were supposed to hide these, but the old compiler does not recognize them):
```
:; ./configure --with-all=auto --with-dev --with-docs=auto
checking for grep that handles long lines and -e... /usr/tgcware/bin/ggrep
checking for egrep... /usr/tgcware/bin/ggrep -E
checking for CONFIG_FLAGS... --with-all=auto --with-dev --with-docs=auto
Network UPS Tools version 2.8.4.166.38 (2.8.4.166.38-204+g55a289784) development iteration
checking build system type... i386-pc-solaris2.8
checking host system type... i386-pc-solaris2.8
checking target system type... i386-pc-solaris2.8
...
```

For posterity, compiler (`gcc (GCC) 4.2.4 (tgcware 4.2.4-2)`) warnings were:
```
libshut.c: In function 'shut_control_msg':
libshut.c:1211: warning: comparison is always false due to limited range of data type
libshut.c:1212: warning: comparison is always false due to limited range of data type

mge-hid.c:253: warning: conflicting types for built-in function 'round'

libhid.c: In function 'refresh_report_buffer':
libhid.c:200: warning: comparison is always false due to limited range of data type
libhid.c: In function 'set_item_buffered':
libhid.c:314: warning: comparison is always false due to limited range of data type
libhid.c: In function 'HIDGetIndexString':
libhid.c:588: warning: comparison is always false due to limited range of data type
libhid.c:600: warning: comparison is always false due to limited range of data type
libhid.c: In function 'HIDGetEvents':
libhid.c:733: warning: comparison is always false due to limited range of data type

hidparser.c: In function 'Parse_ReportDesc':
hidparser.c:705: warning: comparison is always false due to limited range of data type

metasys.c: In function 'upsdrv_updateinfo':
metasys.c:657: warning: implicit declaration of function 'fabsf'
metasys.c:657: warning: incompatible implicit declaration of built-in function 'fabsf'
metasys.c:658: warning: incompatible implicit declaration of built-in function 'fabsf'

powerp-txt.c: In function 'powpan_status':
powerp-txt.c:419: warning: implicit declaration of function 'strtof'

upscode2.c: In function 'upsdrv_updateinfo':
upscode2.c:846: warning: implicit declaration of function 'fabsf'
upscode2.c:846: warning: incompatible implicit declaration of built-in function 'fabsf'

In file included from liebert-gxe.c:24:
ydn23.h:138: warning: #pragma pack(push[, id], <n>) is not supported on this target
ydn23.h:151: warning: #pragma pack(pop[, id], <n>) is not supported on this target

bcmxcp_ser.c: In function 'pw_comm_setup':
bcmxcp_ser.c:338: warning: comparison is always true due to limited range of data type
bcmxcp_ser.c: In function 'pw_comm_setup':
bcmxcp_ser.c:338: warning: comparison is always true due to limited range of data type
```

jimklimov, 2025-10-01:
Following up with a build closer to the system tools:
````
:; PATH="/usr/lib/ccache:/usr/bin:/usr/tgcware/gcc42/bin:/usr/tgcware/i386-pc-solaris2.8/bin:${HOME}/tgcware-minimal/bin"
:; [ -n "$LD_LIBRARY_PATH" ] || LD_LIBRARY_PATH="/usr/lib"
:; LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/usr/tgcware/lib:/usr/tgcware/gcc42/lib:/usr/tgcware/i386-pc-solaris2.8/lib"
:; CONFIG_SHELL=/usr/tgcware/bin/bash
:; export CONFIG_SHELL PATH LD_LIBRARY_PATH
````

The `${HOME}/tgcware-minimal/bin` has curated symlinks for TGCWARE GNU autotools, GNU m4 and GNU make (lacking any other make in this VM), and python2.7:
```
:; ls -la
total 42
drwxr-xr-x   2 jim      other        512 Oct  1 14:28 .
drwxr-xr-x   3 jim      other        512 Oct  1 14:11 ..
lrwxrwxrwx   1 jim      other         38 Oct  1 14:12 aclocal -> ../../../../../usr/tgcware/bin/aclocal
lrwxrwxrwx   1 jim      other         43 Oct  1 14:12 aclocal-1.16 -> ../../../../../usr/tgcware/bin/aclocal-1.16
lrwxrwxrwx   1 jim      other         39 Oct  1 14:11 autoconf -> ../../../../../usr/tgcware/bin/autoconf
lrwxrwxrwx   1 jim      other         41 Oct  1 14:11 autoheader -> ../../../../../usr/tgcware/bin/autoheader
lrwxrwxrwx   1 jim      other         39 Oct  1 14:11 autom4te -> ../../../../../usr/tgcware/bin/autom4te
lrwxrwxrwx   1 jim      other         39 Oct  1 14:11 automake -> ../../../../../usr/tgcware/bin/automake
lrwxrwxrwx   1 jim      other         44 Oct  1 14:11 automake-1.16 -> ../../../../../usr/tgcware/bin/automake-1.16
lrwxrwxrwx   1 jim      other         41 Oct  1 14:11 autoreconf -> ../../../../../usr/tgcware/bin/autoreconf
lrwxrwxrwx   1 jim      other         39 Oct  1 14:11 autoscan -> ../../../../../usr/tgcware/bin/autoscan
lrwxrwxrwx   1 jim      other         41 Oct  1 14:11 autoupdate -> ../../../../../usr/tgcware/bin/autoupdate
lrwxrwxrwx   1 jim      other         34 Oct  1 14:28 gm4 -> ../../../../../usr/tgcware/bin/gm4
lrwxrwxrwx   1 jim      other         35 Oct  1 15:05 gmake -> ../../../../../usr/tgcware/bin/gmake
lrwxrwxrwx   1 jim      other         38 Oct  1 14:13 libtool -> ../../../../../usr/tgcware/bin/libtool
lrwxrwxrwx   1 jim      other         41 Oct  1 14:13 libtoolize -> ../../../../../usr/tgcware/bin/libtoolize
lrwxrwxrwx   1 jim      other         39 Oct  1 14:11 pydoc2.7 -> ../../../../../usr/tgcware/bin/pydoc2.7
lrwxrwxrwx   1 jim      other         46 Oct  1 14:11 pygettext2.7.py -> ../../../../../usr/tgcware/bin/pygettext2.7.py
lrwxrwxrwx   1 jim      other         40 Oct  1 14:11 pynche2.7 -> ../../../../../usr/tgcware/bin/pynche2.7
lrwxrwxrwx   1 jim      other         40 Oct  1 14:11 python2.7 -> ../../../../../usr/tgcware/bin/python2.7
lrwxrwxrwx   1 jim      other         47 Oct  1 14:11 python2.7-config -> ../../../../../usr/tgcware/bin/python2.7-config
```

And so it goes (picked XPG grep despite the PATH though, maybe that's the preference of TGC autotools build?):
```
:; ./configure --with-all=auto --with-dev --with-docs=auto
checking for grep that handles long lines and -e... /usr/xpg4/bin/grep
checking for egrep... /usr/xpg4/bin/grep -E
checking for CONFIG_FLAGS... --with-all=auto --with-dev --with-docs=auto
Network UPS Tools version 2.8.4.166.40 (2.8.4.166.40-206+gd6689771c) development iteration
...

NUT Configuration summary:
==========================

* configured version:   2.8.4.166.40 (2.8.4.166.40-206+gd6689771c) development iteration
* Documentation website base URL:       https://www.networkupstools.org
* enable SSL support:   yes (OpenSSL)
* enable Avahi support: no
* enable libwrap (tcp-wrappers) support:        no
* enable libltdl (Libtool dlopen abstraction) support:  yes
* use default  Python  interpreter:     /home/jim/tgcware-minimal/bin/python2.7
* use specific Python2 interpreter:     /home/jim/tgcware-minimal/bin/python2.7
* use specific Python3 interpreter:
* requested to spell-check the documentation:   no
* requested to build and install documentation: 'auto' => 'man=auto html-single=auto html-chunked=auto pdf=auto'
* would build specific documentation format(s): no
* can install pre-built documentation (release/dist tarball):   no
* would generate ChangeLog* file formats:       text:no adoc:no html-single:no html-chunked:no pdf:no starting from 'v2.8.4' to 'HEAD'
* build and install the development files:      yes
* consider basic SMF support:   auto
* consider basic systemd support:       no
* build with tighter systemd support:   no
* build C++11 codebase (client library, etc.):  no
* build C++ tests with CPPUNIT: no
* User to run as:       nobody
* Group of user to run as:      nobody
[... 46 lines cut ...]

* Compact version of C compiler:        gcc (GCC) 4.2.4 (tgcware 4.2.4-2)
* Compact version of C++ compiler:      g++ (GCC) 4.2.4 (tgcware 4.2.4-2)
* Compact version of C preprocessor:    gcc (GCC) 4.2.4 (tgcware 4.2.4-2)
* host env spec we run on:      i386-pc-solaris2.8
* host env spec we built on:    i386-pc-solaris2.8
* host env spec we built for:   i386-pc-solaris2.8
* host OS short spec we run on: i386-solaris2.8
* host OS short spec we built on:       i386-solaris2.8
* host OS short spec we built for:      i386-solaris2.8
* ccache namespace tag (if ccache is used and new enough):      nut:i386-pc-solaris2.8

NUT Compiler settings:
----------------------

* CC            : gcc
* CFLAGS        : -g -O2 -std=gnu99 -D_REENTRANT -Wno-error
* CXX           : g++
* CXXFLAGS      : -g -O2 -Wno-error
* CPP           : gcc -E
* CPPFLAGS      :
* LD            : /usr/ccs/bin/ld
* LDFLAGS       :
* CONFIG_FLAGS  : --with-all=auto --with-dev --with-docs=auto MAKE=gmake
```

nbriggs, 2025-10-03:
There are still some issues, or at least misleading reporting, when you actually attempt to install the current code (2.8.4.229).
By the way, you can't `pkgadd` a gzipped data stream, so it's not useful for `make package` to compress it.

```
# pkgadd -d NUT_solaris_sparc_package-2.8.4.229.local

The following packages are available:
  1  NUT     Network UPS Tools
             (sparc) 2.8.4.229

Select package(s) you wish to process (or 'all' to process
all packages). (default: all) [?,??,q]: all

Processing package instance <NUT> from </export/home/briggs/nut/NUT_solaris_sparc_package-2.8.4.229.local>

Network UPS Tools(sparc) 2.8.4.229
https://www.networkupstools.org
## Executing checkinstall script.
Checkinstall complete
Using </usr/local/ups> as the package base directory.
## Processing package information.
## Processing system information.
## Verifying disk space requirements.
## Checking for conflicts with packages already installed.

The following files are already installed on the system and are being
used by another package:
* /usr/local/ups/bin <attribute change only>         <<< OK
* /usr/local/ups/etc <attribute change only>          <<< OK

* - conflict with a file which does not belong to any package.

Do you want to install these conflicting files [y,n,?,q] y
## Checking for setuid/setgid programs.

This package contains scripts which will be executed with super-user
permission during the process of installing this package.

Do you want to continue with the installation of <NUT> [y,n,?] y

Installing Network UPS Tools as <NUT>

## Executing preinstall script.
## Installing part 1 of 1.
/usr/local/ups/bin/al175
/usr/local/ups/bin/apcsmart
/usr/local/ups/bin/apcsmart-old
/usr/local/ups/bin/apcupsd-ups
/usr/local/ups/bin/bcmxcp
/usr/local/ups/bin/bcmxcp_usb
/usr/local/ups/bin/belkin
/usr/local/ups/bin/belkinunv
/usr/local/ups/bin/bestfcom
/usr/local/ups/bin/bestfortress
/usr/local/ups/bin/bestuferrups
/usr/local/ups/bin/bestups
/usr/local/ups/bin/bicker_ser
/usr/local/ups/bin/blazer_ser
/usr/local/ups/bin/blazer_usb
/usr/local/ups/bin/clone
/usr/local/ups/bin/clone-outlet
/usr/local/ups/bin/dummy-ups
/usr/local/ups/bin/etapro
/usr/local/ups/bin/everups
/usr/local/ups/bin/failover
/usr/local/ups/bin/gamatronic
/usr/local/ups/bin/genericups
/usr/local/ups/bin/isbmex
/usr/local/ups/bin/ivtscd
/usr/local/ups/bin/liebert
[... 229 lines cut ...]
hemups           <<<< yes, this is the only device defined in ups.conf
Friday, October  3, 2025 at  1:21:04 AM GMT : OK: No more changes to reconcile between smf service instances and device configurations in '/usr/local/ups/etc/ups.conf'
Enable NUT drivers (if any)...
INFO: Detected presence of timeout so will limit service management calls by passing the args: 90s
=== The currently defined service units are:
Starting svc:/system/power/nut-driver:hemups ...
svcadm: svc:/system/power/nut-driver:hemups is misconfigured (lacks "restarter" property group).
Post-process clearing services that failed early...
Clearing svc:/system/power/nut-driver:hemups (if it got broken) ...
svcadm: Instance "svc:/system/power/nut-driver:hemups" is not in a maintenance or degraded state.
Enable NUT upsd data server...
Enable NUT upsmon client...
Enable NUT umbrella service...

Installation of <NUT> was successful.
```
It does seem to have brought everything back online, though --

```
online         2025-10-02T21:20:56 svc:/system/power/nut:default
online         2025-10-02T21:21:08 svc:/system/power/nut-driver-enumerator:default
online         2025-10-02T21:21:14 svc:/system/power/nut-driver:hemups
online         2025-10-02T21:21:14 svc:/system/power/nut-server:default
online         2025-10-02T21:21:15 svc:/system/power/nut-monitor:default
```

jimklimov, 2025-10-03:
Gzipped stream takes less space, and you gzcat it to stdin (or /tmp) IIRC

jimklimov, 2025-10-03:
Hm, not sure if I saw or overlooked these before:
> svcadm: svc:/system/power/nut-driver:hemups is misconfigured (lacks "restarter" property group).

An error with `tail` is not expected indeed, will look what gives...

jimklimov, 2025-10-04:
> that part above doesn't look particularly clean

In `scripts/Solaris/postinstall.in` (and `scripts/Solaris/preremove.in`) it goes with explicit debug ``@SBINDIR@/upsdrvctl -DDDDD stop` (maybe a leftover of my testing, but might also be helpful to see which old drivers it stops).

jimklimov, 2025-10-04:
For `tail`, it disliked the `tail -n +/-NUM` syntax.

jimklimov, 2025-10-04:
For the removal of (or first installation to update from) an old NUT version (which may be pre-2.8.x, pre-NDE, and/or did not use NDE for any or all devices) this is deliberate. The script tries to stop `upsdrvsvcctl` first, and legacy `upsdrvctl` afterwards.

Making it quiet does make sense, the toggles appeared after those packaging scripts were last significantly touched :)

nbriggs, 2025-10-04:
Ah, given the situation, I agree.  The UPS attached to the Solaris 11 box got tested yesterday as part of an annual emergency generator maintenance/test which involved testing the grid/generator transfer switch, and the NUT code dutifully reported the UPS on battery.

jimklimov, 2025-10-08:
@nbriggs : does the package install/update look better on stock Solaris now?

> ```
> # pkgadd -d NUT_solaris_sparc_package-2.8.4.229.local
> 
> The following packages are available:
>   1  NUT     Network UPS Tools
>              (sparc) 2.8.4.229
> ```

One other point that surprised me in your experiments is that it generated a `NUT` package. I think at some point I coded it to make a `NUTx` on Solaris (to follow SVR4 packaging conventions), if it detected a 64-bit build. With `sparc` (not `sparcv9`) mentioned here, maybe it did get built as 32-bit binaries? Can you double-check that (e.g. with `ldd`)? I suppose if you did not `configure` the `CFLAGS`, `CXXFLAGS`, `LDFLAGS` to include `-m64` explicitly, it used whatever the compiler default on the platform was.

Also regarding `--with-solaris-pkg-ips=yes` noted earlier, indeed, the IPS packaging ability remained a placeholder so far... I did not really figure out how to make a locally-usable recipe that does not refer to a particular distribution's dependency package names (or how to reliably detect those names to bake them in - an horde of `pkg` searches for a somehow created list of program/header/library names?). And for SVR4 there was an earlier boilerplate that I expanded (although now I wonder if it also covers all files delivered newly with recent NUT releases, or has to be bumped?)

I did summarize my experiments as a recipe for specifically OpenIndiana distribution at https://github.com/OpenIndiana/oi-userland/tree/oi/hipster/components/sysutils/nut (also a few NUT releases behind, gotta bump it - many of the recipe changes in the past couple of years, like configurable man page sections, were my prerequisites before going to simplifying that packaging recipe), and have yet to figure out how to make an OmniOS one.

CC again @alanc - if you BTW want to snatch that IPS recipe from OpenIndian for Solaris proper, I suppose CDDL licensing allows that? (Maybe after I simplify and bump it for 2.8.4 or later release though)